victoriarobotics/victoria_platform #2: Code to handle ROS navigation messages on Arduinio Teensy

markwomack, 2017-02-14:
Code needs to handle incoming (subscribed) messages related to motor control and outgoing (published) motor encoder information.

griswaldbrooks, 2017-02-14:
You can checkout the topics the Teensy should be publishing by reviewing these tutorials:
http://wiki.ros.org/navigation/Tutorials/RobotSetup
http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom

And you can look at the neato_node for a practical demo.
https://github.com/SV-ROS/intro_to_ros/blob/master/bv80bot/neato_robot/neato_node/nodes/neato.py

This is done in python on a raspberry pi receiving data from the neato, but as you've got ros_serial on the Teensy, that code would live there.
As Mike has pointed out, the Teensy should be publishing the:

- tf transform from base_footprint to odom
- nav_msgs/odom
- sensor_msgs/imu

and receiving
- cmd_vel

As Mike also points out, and other messages that are useful should be published to be used at a later time. The battery one is a very good idea and the ros side of things can know the battery level, as well as when we install the dead man.

griswaldbrooks, 2017-03-07:
For `cmd_vel` you're getting a `geometry_msgs::Twist` message. The two pertinent member variables are `cmd_vel.linear.x` and `cmd_vel.angular.z`, let's call these **vx** and **wz** for now.

To command the motors, let's assume we're going to command each wheel to some linear velocity,
**vl** and **vr** for left and right wheel velocity.
The formulas are then:
**vl = vx - wz * TRACK_RADIUS**
**vl = vx + wz * TRACK_RADIUS**

**TRACK_RADIUS** is the distance between the wheel centers, divided by two. Akin to the robot radius on a round robot. On victoria, this is about 0.235 meters.

To convert linear wheel velocity to angular velocity:
**w_wheel = v_wheel / wheel_radius**
The wheel radius on victoria is about 5 inches, or 0.127 meters.

The top speed of the motors is 45 rpm or 4.71 rad/sec.
Therefore, to get the percentage needed to command the motor:
** w_wheel / w_max **

Note: This is where a PID loop would get inserted.

griswaldbrooks, 2017-03-07:
Servocity quotes these values for the encoder on the motor we're using:

| Cycles Per Revolution (Output Shaft) | Countable Events Per Revolution (Output Shaft) |
|-----------------------------------------|------------------------------------------------------|
| 2,268                                                  | 9,072                                                                     |

I'm not sure which one is the one we want, but one of these will equate to one full wheel rotation.
So, at some regular rate, we'll sample the encoder accumulated values to produce an angular velocity estimate **w_wheel** for each wheel.
Preferably, this would be done using an ISR or some other timed function, though I would still recommend reading the elapsed time rather than relying on the timing to be right.

Once we have that **w_wheel** we use:
**v_wheel = w_wheel * WHEEL_RADIUS**
To get the linear velocity of each wheel.
Then:
**vx = (vl + vr) / 2**
**wz = (vr - vl) / (2 * TRACK_RADIUS)**

After that, http://wiki.ros.org/navigation/Tutorials/RobotSetup/Odom shows the code that should be written, though at a rate much higher than 1 Hz, maybe 100 Hz at least.

Some finer points about computing **w_wheel**.
With these motors, if we want to be able to detect a 1 degree change in the wheels at top speed, we need to sample the encoder at at least 270 Hz, so probably 300 Hz to make it an easier number.
A problem with this sampling is that we're essentially taking the derivative of the accumulated encoder values. Taking the derivative always amplifies noise. Passing this up the chain can cause problems. Imagine these "vibrations" causing the robot to "shake away" in some cases.
@jrcapriles discussed a scheme where we would do some logic to wait until a minimum number of ticks had been accumulated but if too much time passes assume zero velocity.
A better approach I think, is to take this noisy velocity estimate and pass it through a low pass filter, before using it for calculations.
We can know what's called the bandwidth of the motor to tune cutoff frequencies, but essentially you can think of this as just taking an average to smooth things out.

markwomack, 2017-03-08:
Thanks, Gris.  I am going to be looking at this code later this week, and I
will see what I can do.  Do you know of any kind of tutorial on what we are
actually doing here?  I'd like to understand the background better.  I get
the encoder count tells you the distance traveled based on the
radius/circumference, etc, but all this linear and angular velocity...I
need a better grounding and understanding.  Even if there is a basic
robotics book I can buy, that would be good.  Otherwise I am just a monkey.

-Mark

griswaldbrooks, 2017-03-08:
I often recommend this book as a good survey book:
https://www.amazon.com/Computational-Principles-Mobile-Robotics-Gregory/dp/0521692121
I have a copy if you want to borrow it.

I'm not really certain where to point you online without confusing you more.
I don't really know your math and physics background.
This paper has all of the elements that I'm talking about:
http://users.isr.ist.utl.pt/~mir/cadeiras/robmovel/Kinematics.pdf
But I'm not sure how enlightening it is.

Also, a simpler thing to know about is the wheel linear velocity and robot circular motion is all based on this assumption:
https://en.wikipedia.org/wiki/Rolling#Physics_of_simple_rolling
Which we know is wrong, but it's close enough for most things and more complicated models are often times what we call unobservable, though more correct.
You'll find in robots that often people use many simplified models to do things and then correct those assumptions through other means.

griswaldbrooks, 2017-03-17:
I'm going to say this is implemented. The implementation may have issues, which we will open new tickets against.

closed 2017-03-17 by griswaldbrooks