victoriarobotics/victoria_platform #7: Use rosparam defaults

griswaldbrooks, 2017-03-17:
@markwomack 
Instead of doing
https://github.com/victoriarobotics/victoria_platform/blob/master/Arduino/victoria_teensy_ros_node/victoria_teensy_ros_node.ino#L187

Use the ros param defaults.
http://wiki.ros.org/rosserial/Overview/Parameters#CA-80097c1f2a5ef7735d28c0b3f247ceedac26fda6_1

markwomack, 2017-03-17:
The victoria code is doing what is done on the page you pointed me at (I think). It sets a default value if the param cannot be read from ros.  Is there a different method you are suggesting.  I searched that page for "default" and I don't see anything.

griswaldbrooks, 2017-03-17:
Sorry, I read too quickly. I presumed this
http://wiki.ros.org/roscpp_tutorials/Tutorials/Parameters#CA-43555168a8c1f5a1af75766fc207994988094df4_2

Which I'm guessing is not available with rosserial.
It allows you to not have to write the `if` statement

griswaldbrooks, 2017-03-17:
Actually, I don't know that the Teensy version wouldn't allow this. Can you check on that?

markwomack, 2017-03-17:
It looks like the ros_client NodeHandle class does not support it.

http://docs.ros.org/kinetic/api/rosserial_client/html/classros_1_1NodeHandle__.html

Which is unfortunate. It would avoid a lot of boilerplate.  I'll keep looking though.

griswaldbrooks, 2017-03-17:
> classros_1_1NodeHandle__.html

😞

closed 2017-03-20 by griswaldbrooks