victoriarobotics/victoria_platform #8: Wire parameters

griswaldbrooks, 2017-03-17:
@markwomack 
I can't easily find out what these parameters are
https://github.com/victoriarobotics/victoria_platform/blob/master/Arduino/victoria_teensy_ros_node/victoria_teensy_ros_node.ino#L146

Can we make the magic numbers `const int` with names like `I2C_MASTER_ADDRESS` or whatever they are.

markwomack, 2017-03-22:
These are coming from header files from the libraries we are using.  Not sure I can easily do this.

griswaldbrooks, 2017-03-22:
https://github.com/victoriarobotics/victoria_platform/blob/master/Arduino/victoria_teensy_ros_node/victoria_teensy_ros_node.ino#L121

What do you mean?
Why can't we write:
```
// Setup basic I2C master mode pins 18/19, external pullups, 400kHz, 200ms default timeout
const int I2C_WHATEVER_THIS_IS = 0x00;
const int I2C_BUS_SPEED = 400000;
  Wire.begin(I2C_MASTER, I2C_WHATEVER_THIS_IS, I2C_PINS_18_19, I2C_PULLUP_EXT, I2C_BUS_SPEED);
```

markwomack, 2017-03-22:
Yeah, we could, but then we are disconnecting ourselves from the library we are trying to use. I am not sure that is a great strategy. Not like we are using more meaningful names that are adding much, and the comment right above is more informative anyways.

griswaldbrooks, 2017-03-22:
What do you mean by 
>but then we are disconnecting ourselves from the library we are trying to use

markwomack, 2017-03-23:
Oh, you are referring to the values that don't already have #defines.  Now this makes more sense.  The original description was not clear, I thought you were referring to all the parameters.

griswaldbrooks, 2017-03-23:
You're silly @markwomack 
Yes, I was referring to those two magic numbers.

markwomack, 2017-03-31:
#38

closed 2017-03-31 by griswaldbrooks