victoriarobotics/victoria_platform #20: Allow rosparams to be updated dynamically

markwomack, 2017-03-20:
Right now, rosparam values are only loaded in the setup() method.  Need to find a mechanism to update those values without requiring a restart of the Teensy.  Maybe some kind of "update" or "reset" message from ros?

griswaldbrooks, 2017-03-20:
I noticed in `Timer.h` there is a `int stop(int id)` method.
To do reconfigure, I figure it would have to stop the old service, and launch a new one with the updated frequency,

markwomack, 2017-03-20:
Yep.  Just need to keep track of the id it returns when registering the callback (that we are not currently storing). Very doable. Probably need to build a lightweight class around it to keep things simple in the main code. The key is knowing when to do the reset/update.

markwomack, 2017-03-23:
Is there an agreed upon or best practice for communicating a "reset" in ros?

griswaldbrooks, 2017-03-23:
Not sure. Reaching out to friend.

griswaldbrooks, 2017-03-23:
Well, in this case, we're not "resetting", we're changing a parameter.
As we don't want to have the Teensy, continuously polling the parameter server, I'm guessing the best way is to have the Teensy publish a service, that someone can call, to change the parameters.

markwomack, 2017-03-23:
Well, I was just thinking that there would be a topic for "update
parameters" that the teensy would listen to instead of a service. Haven't
done a service before, but we can look into it.

griswaldbrooks, 2017-03-23:
My understanding is that a service would be better because then the Teensy
wouldn't have to poll. I've never done a service either but there are
plenty of tutorials.

markwomack, 2017-03-31:
Did we decide that ros params can only be set once?  Do we want to move towards a service to set them dynamically (and not use ros param)?

griswaldbrooks, 2017-03-31:
ROS params are only set once.
I think we said that there were more pressing issues atm.

closed 2017-04-03 by griswaldbrooks