victoriarobotics/victoria_platform #48: Handle encoder overflow.

markwomack, 2017-04-12:
In doReadEncoders() is probably the ideal place to handle it.

markwomack, 2017-04-14:
I have been running some tests on my TReX rig, to see what happens when the encoder count overflows from both MAX_LONG to MIN_LONG and vica versa (the encoder lib allows a value to be written, so I don't have to wait forever to set up the overflow).  I don't see a problem.  If we subtract the previous reading from the new reading, the difference will always be correct across the boundary.  Am I missing something here?

griswaldbrooks, 2017-04-14:
Wouldn't subtracting the previous from the new cause a problem in the
presence of an underflow and overflow?

markwomack, 2017-04-14:
Thus my confusion.